StealthChesnut/HA-FoxESS-Modbus #2: Identify grid use and export

StealthChesnut, 2022-04-30:
Nice, what about grid use vs export so we can drive automation when we're exporting?

_Originally posted by @williamjeccles in https://github.com/StealthChesnut/HA-FoxESS-Modbus/issues/1#issuecomment-1112421122_

bsg-marvin, 2022-04-30:
> Nice, what about grid use vs export so we can drive automation when we're exporting?
> 
> _Originally posted by @williamjeccles in [#1 (comment)](https://github.com/StealthChesnut/HA-FoxESS-Modbus/issues/1#issuecomment-1112421122)_

For the time being I have solved it (I believe I did) this way:

alias: Car charger solar
description: 'Turn on car charger when battery is full'
trigger:
  - platform: numeric_state
    entity_id: sensor.battery_charge_amount
    for:
      hours: 0
      minutes: 5
      seconds: 0
    above: '18'
    attribute: unit_of_measurement
condition:
  - condition: time
    after: '07:00:00'
    before: '20:00:00'
    weekday:
      - mon
      - tue
      - wed
      - thu
      - fri
      - sat
      - sun
action:
  - type: turn_on
    device_id: 3c467032c727d5295c863566574fb2ea
    entity_id: switch.car
    domain: switch
  - delay:
      hours: 1
      minutes: 0
      seconds: 0
      milliseconds: 0
  - type: turn_off
    device_id: 3c467032c727d5295c863566574fb2ea
    entity_id: switch.car
    domain: switch
initial_state: 'on'
mode: single

williamjeccles, 2022-05-03:
Now implemented into main configuration file

closed 2022-05-03 by williamjeccles