StealthChesnut/HA-FoxESS-Modbus #3: Generation Totals

Captain0Kangaroo, 2022-05-03:
Sorry not an issue as such but was trying to get some totals for generation and its not working.

I've set up the following for example but i'm not getting any numbers logged:

`
utility_meter:
    daily_pv1:
        source: sensor.pv1_power
        cycle: daily
#New Sensor
    daily_pv2:
        source: sensor.pv2_power
        cycle: daily

template:
  - sensor:
      - name: "Solar Power Daily"
        unit_of_measurement: "kWh"
        state: "{{ states('sensor.daily_pv1_power') | float + states('sensor.daily_pv2_power') | float }}"

`
I would then do the same for Monthly, Yearly and Total.

Any ideas where i've gone wrong?

Daibutt, 2022-05-03:
Yes looks like you're missing the integration bits.
Power sensors go to an integration sensor (which will just keep adding them up forever)
The intgeration sensors then feed into the utility meters for d, m, y etc.. so you get understandable numbers.
Examples in here;
https://github.com/macxq/foxess-ha/wiki/Understand-PV-string-power-generation-using-foxess-ha

Captain0Kangaroo, 2022-05-03:
> Yes looks like you're missing the integration bits. Power sensors go to an integration sensor (which will just keep adding them up forever) The intgeration sensors then feed into the utility meters [...]

I'm obviously being a dumb ass as i just cant seem to get it working. I figured it out using the HACS integration and have totals running fine but just cant seem to get them working with the hard wired system.

Daibutt, 2022-05-03:
Your utility meters are looking at the power sensors, which won't work. You need some integration sensors like these
(ignore the foxess parts in the names)

platform: integration
   source: sensor.foxess_pv1_power
   name: pv1_sum
   unit_time: h
   round: 2
platform: integration
   source: sensor.foxess_pv2_power
   name: pv2_sum
   unit_time: h
   round: 2

Then you feed the integration sensors into the utilty meters (not the power sensors)

utility_meter:
   daily_pv1_sum:
      source: sensor.pv1_sum
      cycle: daily
   daily_pv2_sum:
      source: sensor.pv2_sum
      cycle: daily

Captain0Kangaroo, 2022-05-03:
I've just tried again with the following:

utility_meter:
  daily_pv1_sum:
    source: sensor.pv1_sum
    cycle: daily
  daily_pv2_sum:
     source: sensor.pv2_sum
     cycle: daily

sensor: 
  - platform: integration
    source: sensor.pv1_power
    name: pv1_sum
    unit_time: h
    round: 2
  - platform: integration
    source: sensor.pv2_power
    name: pv2_sum
    unit_time: h
    round: 2

template:
  - sensor:
      - name: "Solar Power Daily 2"
        unit_of_measurement: "kWh"
        state: "{{ states('sensor.daily_pv1_sum') | float + states('sensor.daily_pv2_sum') | float }}"

But its spitting out an increasing figure which is just growing all the time and is obviously wrong:
[image]

Daibutt, 2022-05-03:
Yes it will keep increasing till midnight then go back to zero. 
Possibly you have W and kW mixed up somewhere. The modbus stuff pulls in Watts I believe, so your Solar power Daily 2 is probably Wh not kWh.

Daibutt, 2022-05-03:
What are these power sensors reading;

williamjeccles, 2022-05-03:
Tested with latest configuration file and all working as expected. Please confirm you're seeing the same using the latest config file?

williamjeccles, 2022-05-04:
Tested on 5 instances of HA v2022.4.7 - no new issues reported

williamjeccles, 2022-05-04:
I think I've discovered the issue, seeing 1,000's of kWh instead of single kWh. Working with Dave on a solution and config update.

williamjeccles, 2022-05-04:
Now resolved in latest updated config and Modbus

closed 2022-05-04 by williamjeccles