StealthChesnut/HA-FoxESS-Modbus #12: Load Power goes negative

StealthChesnut, 2022-05-09:
I have observed the Load Power reading going very low and negative at certain times. This seems to be when the battery SoC is full/99%, good solar generation (over 3kW), and load is less than the solar generation.

Not sure if this is anything to do with the integration config, or just the data that's coming from the inverter. Seeing matching data in the Fox portal, i.e. LoadsPower reading -0.37kW.

williamjeccles, 2022-05-09:
@StealthChesnut does my solution for negative pv values also work for grid use negative values?
https://github.com/StealthChesnut/HA-FoxESS-Modbus/discussions/7

StealthChesnut, 2022-05-11:
I'm not sure at the moment. That feels like it's masking the problem rather than trying to find out the cause. I don't think it's the integration, given both the data in HA and the fox cloud say the same thing - my house load goes negative. It might be that the inverter just can't figure out what's going on / the maths is wrong.

This might be a "can't fix".

Daibutt, 2022-05-11:
Also shouldn't feedin match the shelley? Maybe you have a dodgy CT on the Fox inverter?

StealthChesnut, 2022-05-12:
I don't think this is the integration - likely wiring or the clamp as you say. The figures here are also reported in the cloud.

I would expect feedin to match, except for the losses in the system which don't seem to get accounted for. To make the numbers all line up you'd need to add those in as well.

williamjeccles, 2022-05-29:
Have been monitoring for a few weeks now, all appears stable. Going to close down this issue unless we see it again.

closed 2022-05-29 by williamjeccles

StealthChesnut, 2022-05-30:
My installer has tried to blame this on the Fox app, saying it puts out bad numbers sometimes. 🙄 

We'll close this as I think it's down to my install, and I can see a long battle ahead with the installer to get to the root cause.