StealthChesnut/HA-FoxESS-Modbus #14: Battery charge wrong when force charging

welshdave, 2022-05-09:
When force charging I'm getting inflated values for battery charge. I only have a 7.8kWh battery, and checking with my meter data confirms that I didn't use anywhere near this much electricity from the grid.

[image]

williamjeccles, 2022-05-09:
@welshdave could you show a chart of the Battery_Charge_Daily for the timespan of the issue above?

welshdave, 2022-05-09:
This is battery charge daily:

[image]

It doesn't seem to marry up with the data from sensor.battery_charge:

[image]

Daibutt, 2022-05-09:
@williamjeccles I see the same with battery discharge
[image]
That 6kWh jump was when I put a 3kW kettle on for 3 minutes! Work that out lol.
I think it may possibly be down to the way the templates work for feedin/grid consumption and bat charge/discharge. The logic sets them to **exactly zero**, sometimes for many hours, which seems to be messing with the integration sensors?
Most "normal" sensors have a bit of noise on them so we don't see this effect usually.
See my comment on this closed out issue, where I have created a random noise sensor to use instead of exactly zero.
https://github.com/StealthChesnut/HA-FoxESS-Modbus/issues/15#issuecomment-1121151086

Still early days but I haven't seen a similar jump since. Will keep monitoring.

welshdave, 2022-05-09:
@Daibutt Interesting, I'll give that a go and see what it does.

Daibutt, 2022-05-09:
> @Daibutt Interesting, I'll give that a go and see what it does.

Great, I'm interested to see if that works for you. Only had mine running for 12hrs or so but still looking OK so far.

welshdave, 2022-05-10:
Looking a lot better this morning:

[image]

Captain0Kangaroo, 2022-05-10:
I didn't charge last night (although I wish I did as I'm now pulling from the gird 🙄)

However I'm seeing a big spike in my battery charge figure which is incorrect.

Dave - did your solution fix this?

[image]

Daibutt, 2022-05-10:
@Captain0Kangaroo Mine looks good so far, no big spikes in charge/discharge or feeedin/consumption

Captain0Kangaroo, 2022-05-10:
> @Captain0Kangaroo Mine looks good so far, no big spikes in charge/discharge or feeedin/consumption
> 
> [image]

Could you paste your code please. I've tried to copy it but doesn't paste well and when i typed it in i think i've got a mistake somewhere as when doing a Check Configuration it throws back lots of errors.

Thanks

Daibutt, 2022-05-10:
OK... add this in your sensor part of configuration.yaml (mine is just below the integration sensors for bat discharge, grid consumption etc.).;

```
- platform: random
  name: random_watts
  minimum: 1
  maximum: 10
```

Then find your template for battery charge
[image]

and replace the zero indicated with this code;

`{{ (states('sensor.random_watts') | float - 5.5 ) / 1000 }}`

first curly bracket needs to line up with the others in the template so it looks like this;

[image]

welshdave, 2022-05-10:
@Captain0Kangaroo - the changes @Daibutt suggested worked to stop spikes for me last night, I'll be keeping an eye on it for the next few days now.

Captain0Kangaroo, 2022-05-10:
> OK... add this in your sensor part of configuration.yaml

Nice one - Thank you...

Captain0Kangaroo, 2022-05-10:
Not sure what's happening now but my battery discharge and feedin are the same figure and change exactly with each other.

[image]

williamjeccles, 2022-05-11:
@Daibutt I think using the sum vs daily for the energy meters will solve this, doesn't it?

williamjeccles, 2022-05-11:
correction, I see the same issue with sum. going to monitor and if still an issue try your solution.

StealthChesnut, 2022-05-11:
I'm looking into an alternative method by adding "method: left" to the Integration sensors. The documentation for them shows the default method to be Trapezoidal, and recommend Left for "spiky" loads. I suspect it won't work though.

@Daibutt is right in that if the value is just 0 for 12 hours, then goes to 2kW, the two data points are 0 and 2, so it'll then try to average that out coming in at (very rough guess) 2ish kWh instead of 0.

I think this would likely work just as well if we had alternating 0.0001W and -0.0001W fed into the integrations every minute or so that at any point they're only integrating over 60s or less, and not time gaps of hours. It's not the fact it's random noise being put in, it's just getting Home Assistant to log additional data points more frequently, therefore shortening the time that the integration is averaging values for.

Easy version: in the wikipedia article (https://en.wikipedia.org/wiki/Riemann_sum) graphics, our bars are too wide. If we use Daibutt's method we make the bars much narrower, making the averaging much higher resolution.

Daibutt, 2022-05-11:
Agree, alternating +1W - 1W would do it too as long as the integration
sensor sees something changing.
I just used the random sensor as it was an easy way to generate a
constantly changing number.

StealthChesnut, 2022-05-13:
Since adding method: left to my integration definitions on the evening of the 10th I've seen less jumps, and also lower reported usage (this is bat_charge_daily):
[image]

Are you still having success @Daibutt ? I am thinking I will add the method: left property, but also put in your random noise sensor commented out so that people can use it easily if they wish?

Daibutt, 2022-05-13:
@StealthChesnut Yes Chris,  the random noise sensor is still working well for me. 
[image]

Glad you got the method:left bit to work. I thought I'd tried that at some point and it didn't fix the issue, but maybe I tried right and did't get round to left !

Yes, if you don't think it will cause confusion, by all means include the random noise stuff commented out.

StealthChesnut, 2022-05-13:
Added method: left to integration sensors, closing this issue. Will add a noise sensor as option.

closed 2022-05-13 by StealthChesnut