StealthChesnut/HA-FoxESS-Modbus #26: Reinvestigate the LAN Port on inverter

StealthChesnut, 2022-06-09:
### Discussed in https://github.com/StealthChesnut/HA-FoxESS-Modbus/discussions/25

<div type='discussions-op-text'>

<sup>Originally posted by **mdt3k** June  8, 2022</sup>
My H1-5.0-E inverter has a LAN port, in addition to the RS485 port that the WiFi logger dongle connects to.

I decided to connect a computer to it, and discovered the port hosts a ModbusTCP server. I doesn't appear fussy about what equipment address you poke it with (I tried 1 and 247), it will respond to requests for modbus input registers 300-399, and holding registers 30,000-30,027, 31,000-31,038 and 44,000-44,002 (it is possible it will respond to other registers, I only tried every 100th register). The server responds straight away, and the solar-related values change at least every 5 seconds (I have not logged data for a long period at a faster rate than this, when register scanning I queried every 0.25 seconds and this didn't seem to cause any problems). It is interesting (to me at least) that these are not the same register numbers that everyone else is reporting, and they do not seem to contain exactly the same information as available via the WiFi dongle. I have not yet worked up the enthusiasm to create an RS485 break out box, but for now I assume my dongle is working the same as everyone elses.

The registers return the following information:
- 300-399: the values 1-11 and then zeros. I am unclear on the purpose of these registers
- 30,0000-30,027: 16 characters consisting of the model number (H1-5.0-E in my case) followed by some spaces to make up the 16 characters; then the firmware version numbers (138, 102 and 129 in my case), and then some zeros
- 31000-31038: Solar-related information (see decode below). I don't currently have a battery connected so I expect some of the registers I haven't decoded are related to that.
- 44,000-44,002: So far these have always contained the values 0; 0; 25,704 (=0x6468 or b0110 0100 0110 1000). It's not clear to me what this is, I would guess configuration settings, but I have not beeen able to map it on to any of the tables in the various manuals I have found floating around on the internet. I am interested in whether anyone can shed any light on this. Perhaps someone has the correct protocol manual for my inverter?

My decode of registers 31,000-31,038:
All the values appear to be signed, and where a multiplier is listed they are fixed point values (x10 means a reported value of 2 should be interpreted as 0.2, x100 means 2 -> 0.02)
31000	PV1 Voltage x10
31001	PV1 Current x10
31002	PV1 Watts
31003	PV2 Voltage x10
31004	PV2 Current x10
31005	PV2 Watts
31006	Grid Voltage x10
31007	Generated AC Current x10
31008	Generated AC WattsCurrent x10 
31009	Grid Frequency Hz x100
31010	unknown_1 
31011	unknown_2 
31012	unknown_3 
31013	unknown_4 
31014	Export Watts (negative = importing)
31015	unknown_5 
31016	House Usage Watts
31017	unknown_6 
31018	Ambient Temperature degrees C x10
31019	Inverter Temperature degrees C x10 (the fox cloud has these values transposed, but that makes no sense)
31020	Inverter Battery Voltage x10
31021	Inverter Battery Current x10
31022	unknown_8 
31023	unknown_9 
31024	unknown_10 
31025	unknown_11 
31026	unknown_12 
31027	Inverter State (0 = Standby, 2 = On Grid, other values = ?)
31028	Possibly the “Inverter Mode” from the Fox protocol manuals
31029	Possibly the CT meter type  from the Fox protocol manuals
31030	unknown_13 
31031	unknown_14 
31032	unknown_15 
31033	unknown_16 
31034	unknown_17 
31035	unknown_18 
31036	unknown_19 
31037	unknown_20 
31038	unknown_21 
(there is no unknown_7 as I worked out what that was).

Can anyone shed any light on any of the unknown values? In particular the use of registers 300+ and 44000+?</div>

StealthChesnut, 2022-07-29:
Progress made, values being read over LAN connection. Will add this as an option and then bring it up to parity with the RS485 option as the registers are confirmed.

djdeant, 2022-07-29:
@StealthChesnut I have discovered that these registers also exist:
31039
40006
Unknown purpose

djdeant, 2022-07-29:
Also 100-115 are input coil registers.  Unknown purposes too.

djdeant, 2022-07-29:
@StealthChesnut 
31024 is Battery SOC.  I will endeavour to work out the rest if you want (my system has batteries connected)

StealthChesnut, 2022-07-29:
The known registers are detailed here:
https://github.com/StealthChesnut/HA-FoxESS-Modbus/wiki/Data-Register-Reference---H1-AC1

djdeant, 2022-07-29:
Ok thanks.  I've finished matching up all my registers and updated my modbus.yaml to use the new ones.

StealthChesnut, 2022-08-04:
This issue has been fixed in the latest release. There's now two options - USB for an RS485 or wifi converter, and LAN for direct connection to the inverter.

closed 2022-08-04 by StealthChesnut