StealthChesnut/HA-FoxESS-Modbus #33: System Losses sensor giving incorrect values (LAN)

StealthChesnut, 2022-08-05:
The system losses sensor appears to be reading more like a current power generation figure, not a losses figure.

Units may be wrong, or source data may be wrong (incorrect register definition in the LAN config?)

StealthChesnut, 2022-08-05:
Likely related to Discussion https://github.com/StealthChesnut/HA-FoxESS-Modbus/discussions/9

TonyM1958, 2023-04-03:
Don't know if this helps, I changed how system_losses are calculated in my setup, using RPower 31008. More details in the code snippet here: https://foxesscommunity.com/viewtopic.php?p=907&sid=9a6298b7f5e2a6a928328633d103530a#p907

StealthChesnut, 2023-04-04:
Thanks Tony - appreciate the contribution. I'm not sure how we'd verify how accurate any approach is. Any thoughts @Daibutt ?

TonyM1958, 2023-04-04:
I've added a comment to the disucssion on the community forum linked above, as the calculations depend on some sensor name changes in your code. I made these changes for consistency between USB and LAN and also to align more with Nathan Marlor. May be worth having a look to see if you want to incorporate any of the changes into the next release?

Ideally, I think the various names would be resolved by adding consistent unique_ids to the entities, so this is something I am also looking into.

TonyM1958, 2023-04-04:
oops - looked at the code I posted and realised I hadn't actually copied the code I use for System Losses! Corrected this now.