StealthChesnut/HA-FoxESS-Modbus #40: T Series (G3) Support ?

3tes, 2022-08-16:
Hello,
i would like to know if anyone had luck with setting up a T Series.
It has a slightly different connector then shown here on the GitHub
this is the connector:
[image]

And if it is supported how do you connect to it?
Is the connector available for purchase or do you just put some cables in it ?

StealthChesnut, 2022-08-17:
You would use pins 1 and 2 to connect, using one of the usb to rs485 converters for hooking up to that connector.

Does the T series have a LAN port? That would be easier to connect to.

3tes, 2022-08-17:
Sadly not it only has a USB port where you can connect a WIFI dongle.
How would I go about connecting to the port ? Just jam some wires into it ?

StealthChesnut, 2022-08-17:
No, there should be a connector or plug that you can wire into, then use that to connect up to the port. 

I would not jam any wires into your inverter.

3tes, 2022-08-17:
Oh thank you didn't se that came with one

3tes, 2022-08-24:
I set it all up but now i get this error 
[image]

Safs, 2022-09-13:
Hello! I try connect T series with Elfin-EW11, bu no luck...
I have FOXESS Modbus Communication Protocol for this series, maybe it is help? 
[MonitoringProtocolVB1.02.00_20220127_ for TG3.pdf](https://github.com/StealthChesnut/HA-FoxESS-Modbus/files/9557651/MonitoringProtocolVB1.02.00_20220127_.for.TG3.pdf)

assembly12, 2022-09-13:
Hi! I just recently got a T Series installed and have been trying to get the data over Modbus into HA.
I haven't had any luck, though (see my efforts up till now at the [HA forum](https://community.home-assistant.io/t/foxess-inverter-modbus/459313/7). Because of this thread I found out that the inverter probably uses a different protocol from the ones supported by this integration.
I can't figure out though, what the correct adresses should be. I've got to say, that I know nothing about modbus at all. So if it's not to much effort, perhaps one of the contributors of this integration could have a look at the protocol.
I'm happy to test things out, but right now I'm just not sure what else I can try.

StealthChesnut, 2022-09-13:
The document suggested by @Safs only describes how the inverter is communicating with the Fox cloud. Unless you intercept this traffic, or capture it into a proxy, this isn't any use for connecting directly to the T series like this integration does to the H1.

@assembly12 There is very little documentation, and without a T series inverter for me to work on I can't make any progress. The process involves sending commands using different settings - all trial and error, which is very time consuming - much like 3tes is trying.

@3tes timeout indicates you're not connecting to the modbus target (inverter). Are you sure the IP, modbus address and connection settings are correct? You may need to check the IP and modbus address (slave number?) on your front panel menu. The connection settings I'm not sure how you'd check those.

Safs, 2022-09-14:
There is answer from foxess support:
“It is the same data. The inverter sends the same data to the server and on the rs485 port.
This is not a traditional modbus. There is no question-answer communication.”

Maybe it helps…

Safs, 2022-09-15:
I connect inverter with Elfin-EW11, Elfin-EW11 configured as MMQT client and get in HA MMQT mosquito messages like this:

```
Message 15 received on fox_modbus at 09:05:
b'~~\x02c"zV\x00\xe8\x00\x00\x04\xf8\x00\x00\t4\x00\x13\x13\x87\x01\x99\t.\x00\x14\x13\x87\x01\xae\t@\x00\x14\x13\x87\x01\xb1\x15\xe3\x00\x0e\x00\x00\r\x18\x00\r\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x1e\x00\x1e\x00#\x00\x05\x00\x00T=\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00)\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xda\x00\x02\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x02\x00\x02\x00\x00\x00\x00\x03\x93\xfd\xd0\x05n\x00\x01\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xff\xff\x00\x02\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x01\x19\x92\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x14]\xe7\xe7'
```
What can i do with it?

StealthChesnut, 2022-09-15:
I can see you're also asking here https://community.home-assistant.io/t/foxess-inverter-modbus/459313/15 

This looks like you are now receiving data from the inverter, and you will need to create some sort of parser to take the individual data blocks and put them into sensors in HA.

Modbus usually works with one device being the master, which sends commands, and the rest of the devices responding. You can't have 2 masters. For the H1 inverters, HA acts as the master and requests specific data. It looks like your inverter behaves differently - I'm guessing that it is the master and is just sending data out.

There were some other projects on github that were looking into that method of working - maybe they would be some help?

Safs, 2022-09-15:
Yes, inventer  just sending data out...

StealthChesnut, 2022-10-03:
@assembly12 are you listening for the data that the inverter is sending out on the RS485 conn, then pushing that to home assistant?

assembly12, 2022-10-03:
Yeah. The inverter pushes one of 4 messages everey 30sec, no need to request it. Only one of them contains the real time information. So the HA sensors update every 2 minutes.
This new protocol does not resemble the modbus protocol one to one. Therefore it's not possible to use the modbus component. Instead I use a custum UART device to listen to the traffic comming in on RX.

StealthChesnut, 2022-10-03:
ok - I'm going to close this issue as it's going to be outside the scope of this project to take on support of the T series, given what you've had to do @assembly12 

I've updated my project's readme to include a link to your project. good work!

closed 2022-10-03 by StealthChesnut