StealthChesnut/HA-FoxESS-Modbus #41: Octopus itegration?

dbrb2, 2022-08-25:
Hi.
Thanks for the excellent work on this. 
I apologise that this is only tangentially related to your code: 

I am trying to get my head around how (if) the integration with Octopus, in terms of when the system is exporting and when it is importing, is managed with the foxESS kit

Does the use of the Modbus interface for local access using your code affect this - or is that managed between the Fox Cloud and the Octopus cloud....or something else? 

Of the many inverters out there, any thoughts on Fox vs any of the others as a good solution? 

Cheers!

spmaid, 2022-08-25:
I’m not sure what integration with Octopus there is other than I think they’ve loosely claims it works with Octopus to charge at a given time a slot. I’m not aware of any way to force export to grid e.g. to take advantage of Agile Export. 

I personally wouldn’t choose FoxESS inverters knowing what I know now but stuck with them for the foreseeable. They’re cheap and they work but hard work to integrate with (although this integration is superb!)

StealthChesnut, 2022-08-25:
At the moment there's no overlap with octopus integrations as modbus is only reading data. There might be conflict if I can write settings, but that will be a major change and flagged up.

There are other integrations that can use the fox site to update settings, but I don't think there's any official fox features that make it easy to use things like agile, etc.

art103, 2022-08-25:
@StealthChesnut, have you had any success writing to the registers?
mbpoll claims success when writing but the values do not change.

Even just the ability to force charge (via ModBus) would be a start and it's possible that switching the inverter to 'feed in' mode might give us some limited control over export too.

I've proved that meter emulation is a valid way to control both battery charge and discharge with fine granularity; however it requires custom HW, which is not ideal.

StealthChesnut, 2022-08-25:
Not as yet, but I've not had a lot of time to try recently. It'll be at
least a week until I can look again unfortunately.

That is my next goal though.