StealthChesnut/HA-FoxESS-Modbus #51: H3 Hybrid Series Support (inverter LAN interface)

benbeton, 2022-09-17:
Hi, currently there is only support for  the H1 Version .

I own a H3-6.0E and tried the integration, but the registers seems to be wrong. I do not get any information.
I can confirm, that Port 502 is open, but that's it.

Maybe the registers are different!? @StealthChesnut can you share some information, how you found the correct registers for the H1 inverter interface?

tagware, 2022-09-18:
Hi All,

You can use gmodbus as this is a test rig to check out the connection.
I've have assumed that you are using a Wired connection? i.e. Cat cable
from the inverter to a Hub/Switch/ADSL router.
https://www.gineers.com/en/products_soft_other_gmodbus.php
<http://qmodbus.sourceforge.net>

You download this zip to a folder. Then use Windows File explore to select
the file and extract 3 files using the top menu. That should reconise it as
a compressed file. Once, complete double click the Gmodbus.exe n folder.

Once opened. Type in the IP address of the Fox Inverter in the "Remote IP"
. Then,  check that the Remote port is set to 502 and Modbus is 237.
Under "Read Modbus"  check the "Holding" is selected, Type in Address 31000
and count is 32. Then, select "Connect TCP" button.

Bottom table should confirm connection is open, then select "Read
Registers". The table to the right should fill with all registers. Then,
you can confirm things are working and look at the wiki link in the
original to look at registers meaning.

The Fox inverter info page will provide the IP given to the inverter. Or,
look on the ADSL/DHCP to view IP

I have copied the Fox part of the "configuration.yaml" . This is my working
out. Use at your own risk. However, it working with HA
It's crude but it works good enough.

Have a nice day as our American cousin would say.

Regards
*David Vincent*
*Mobile:* *07785 114965*

*Website: www.tagware.co.uk <http://www.tagware.co.uk/> *
*Blog:     **blog.tagware.co.uk* <http://tagware.info/>

<http://www.tagware.co.uk/>   *13 Sandersfield Road, **Banstead, **Surrey,**SM7
2DH. * *Registered No: 03006925 England. ** - *   *wacky.battle.blaze
<https://map.what3words.com/wacky.battle.blaze> *

This e-mail contains confidential information and is for the exclusive use
of the addressee's. If you are not the addressee, then any distribution,
copying or use of this e-mail is prohibited. If received in error, please
advise the sender and delete it immediately. We accept no liability for any
loss or damage suffered by any person arising from the use of this e-mail.
How we use your data can be viewed here
<https://tagware.co.uk/privacy-policy/>

# Example configuration.yaml entry
modbus:
  - name: hub1
    type: tcp
    host: 192.168.0.37
    port: 502
    sensors:
      - name: FOX_HZ
        unit_of_measurement: Hz
        slave: 1
        address: 31009
#        data_type: float
        scale: 0.01
        precision: 2
      - name: FOX_INV_TEMP
        unit_of_measurement: C
        slave: 1
        address: 31019
#        data_type: float
        scale: 0.1
        precision: 2
      - name: FOX_Input_Voltage
        unit_of_measurement: V
        slave: 1
        address: 31006
#        data_type: float
        scale: 0.1
        precision: 2
      - name: FOX_Battery-SoC
        unit_of_measurement: "%"
        slave: 1
        address: 31024
#        data_type: float
#        scale: 0.01
#        precision: 2
      - name: FOX_Load-Power
        unit_of_measurement: W
        slave: 1
        address: 31016
#        data_type: float
        scale: 1
#        precision: 2
      - name: FOX_InvBatVolt
        unit_of_measurement: V
        slave: 1
        address: 31020
#        data_type: float
        scale: 0.1
        precision: 2
      - name: FOX_Battery-Charg_Discharge
        unit_of_measurement: W
        slave: 1
        address: 31022
      - name: FOX_InvBatPower
        unit_of_measurement: kW
        slave: 1
        address: 31021
#        data_type: float
        scale: 0.1
        precision: 2
      - name: FOX_GridExport
        unit_of_measurement: kW
        slave: 1
        address: 31014
#        data_type: float
        scale: 0.001
        precision: 2
        
      - name: FOX_InvMode
        unit_of_measurement: Mode
        slave: 1
        address: 31027

benbeton, 2022-09-18:
thank you @tagware, hope I can help to get this woking! did the steps, sent 31000,  but received the following in the log: 

> Trying to connect to remote TCP server...
Connected TCP/IP, at 10:07:09
Send: 03 E8 00 00 00 06 ED 03 79 18 00 20 
Received: ED 83 02 01 04 

Sees to be a hex code, but I have no idea if this is correct, I expected the value from the article!?

here are the settings
[image]

tagware, 2022-09-18:
Ok.

Look like you are connected to something on 502.

Did you push the "Read Register" button?

Should see the table fill and Text in bottom box of

Send: 03 E9 00 00 00 06 ED 03 79 18 00 20
Received: ED 03 40 00 00 00 00 00 00 00 00 00 00 00 00 09 3E 00 11 FE 81 13
8F 00 00 00 00 00 00 00 00 FF FD 00 00 FE 83 00 00 00 CA 01 28 0E B1 FF F6
FE 6E 00 9E 00 3F 00 D2 01 F4 00 02 00 01 00 01 00 00 00 00 DA F2

:o)

Regards
*David Vincent*
*Mobile:* *07785 114965*

*Website: www.tagware.co.uk <http://www.tagware.co.uk/> *
*Blog:     **blog.tagware.co.uk* <http://tagware.info/>

<http://www.tagware.co.uk/>   *13 Sandersfield Road, **Banstead, **Surrey,**SM7
2DH. * *Registered No: 03006925 England. ** - *   *wacky.battle.blaze
<https://map.what3words.com/wacky.battle.blaze> *

This e-mail contains confidential information and is for the exclusive use
of the addressee's. If you are not the addressee, then any distribution,
copying or use of this e-mail is prohibited. If received in error, please
advise the sender and delete it immediately. We accept no liability for any
loss or damage suffered by any person arising from the use of this e-mail.
How we use your data can be viewed here
<https://tagware.co.uk/privacy-policy/>

benbeton, 2022-09-18:
yeah, pushed the button and the only thing that happens is the output above
> Received: ED 83 02 01 04

it does not change, even if I send 31001 e.g.

tagware, 2022-09-18:
Your certain that the IP ? Ping the IP and remove the inverter cat cable to
check. Obviously put it back in again.

Cable coming out of the eithernet plug at bottom of the inverter into the
router / switch ? Might want to pull it out & reset it back in.

Is the inverter talking to the cloud via WiFi?

Well it's working here with those settings.

Mine is an AC1. Have you a firewall on the system trying to communicate?

Running out of ideas.

Let me know when you have had a go at the above. I think some more.

StealthChesnut, 2022-09-18:
So the response code can be broken down to indicate what's going on. See this page for some definitions: https://ipc2u.com/articles/knowledge-base/modbus-rtu-made-simple-with-detailed-descriptions-and-examples/

Quite a way down the page: 
83 indicates an error response to your 03 request
02 indicates an error code
01 function code can't be processed?

It could be the type of read command isn't correct (holding register or not?), Also it looked like you have slave ID 237. It might not matter for TCPIP connections but try 247.

benbeton, 2022-09-19:
after some researching i figured out, that maybe the firmware currently has a modbus implementation, but does function. 
lets's wait a few days, and see if the new firmware arrives and has working implementation.

benbeton, 2022-09-20:
Received the file and updated the inverter today to: 

> Master : 1.31, Slave : 1.02, Manager : 1.43

but still no luck, gmodbus still gives me: 
[image]

and the integration shows:
[image]

do I miss something, has anything to be enabled in the inverter menu? 
could it be possible that the H3 has other registers? How can I start searching for them ?

tagware, 2022-09-20:
Just for clarification

This is what's on my system.

AC1-5.0-E

Try 4000 range?

benbeton, 2022-09-20:
mine is a H3-6.0E, i tries multiple ranges

```
11000
21000
31000
41000
51000
```

I do not have the information, how I can test more, alway's seeing the same output

tagware, 2022-09-20:
Ok.

Straws now.

Is the CAT cable in 10/100Mb or 1Gb nic on the switch?

Try another Cat5 cable.

Does this PC have a active Firewall?

You did say if the Cloud account is reporting correctly?

StealthChesnut, 2022-09-20:
Could also try reading the registers as input, not holding, and also for a count of 1 or 4, not 32. Basically try different settings in the read area. I wouldn't touch the write area!

If you try to read 32 registers and there's only 30 registers you will get an error.

Partovic, 2022-09-27:
I was also considering to buy a Fox ESS H3 series inverter. I asked the "technical support" if they support the modbus protocol.
They answered that they currently don't support the protocol for this series.

benbeton, 2022-10-03:
> I was also considering to buy a Fox ESS H3 series inverter. I asked the "technical support" if they support the modbus protocol.
They answered that they currently don't support the protocol for this series.

do they said on which Interface? only the internal one? I‘m thinking of buying the RS484 to Ethernet Adapter and maybe that will work, instead of the internal Ethernet interface

> FoxESS also don't support any kind of consumer access to their devices, luckily, ways have been found (for some at least)! :-)
can you explain this a bit more detailed?

spmaid, 2022-10-03:
> can you explain this a bit more detailed?

Well, this HA pluggin 😊 thanks to the hard work dedicated to it

rsaemann, 2022-10-09:
Hi, alltogether.
 I have a FoxEss H3-8.0-E Inverter, which does not talk to Modbus at all. Maybe we can share our experience.
I can ping the Inverter, that is connected via the LAN-Adapter (Inverter-Router-Laptop). But all attempts to portscan (PortScan & CAS Modbus Scanner) or directly talk via GModbus show no reaction. G-Modbus gives me a `Trying to connect to remote TCP server...
Socket error #10060`

My Inverter is in island-mode and I blocked the Internet connection of the inverter in my router (AVM). It has never initially connected to the cloud. Maybe there is something like an initialization via cloud before it can speak locally via ModBus?

Softwareversion is:
Master 1.25
Slave 1.02
Comm 1.29

Daibutt, 2022-10-09:
Someone has managed to get this working on an H3.

"Finally a new firmware available and modbus working for AIO-H3 inverters. I've just successfully tested reading of registers using USR-W610/rs485. I have to modify slightly sensor defined in the modbusLAN.yaml. The new firmware installed on my inverter: Master : 1.31, Slave : 1.00, Manager : 1.36"

Here is link to the post, you will need to scroll down quite a bit as there is alot of discussion. the post is from ~ September this year.

https://www.facebook.com/groups/foxessownersgroup/permalink/1581233825592127/

domerich, 2022-12-06:
I have a I have a FoxEss H3-8.0-E Inverter and today I connected Pin 1 and 2 to the Elfin-EW11 Elfin-EW11-0 RS485 to WiFi Module. After starting up the inverter now shows InvRelay error and will not power on. I think I killed it.

is there a ready YAML file for H3 to LAN I could use? 

Thanks
[image]