StealthChesnut/HA-FoxESS-Modbus #60: H3 Register Tables

rsaemann, 2022-10-26:
I have a 3-Phase inverter H3-8.0-E and I want to share my register findings for Modbus over Serial with you.

Input solar strings:
31000    Voltage String 1   "V"   x0.1
31001   Current String 1    "A"   x0.1
31002   Power String 1      "W"  x1
31003    Voltage String 2   "V"   x0.1
31004   Current String 2    "A"   x0.1
31005   Power String 2      "W"  x1

Output to grid/EPS:
31016   Voltage Phase 1   "V" x0.1
31017   Voltage Phase 2   "V" x0.1
31018   Voltage Phase 3   "V" x0.1

31019   Current Phase 1   "A" x0.1
31020   Current Phase 2   "A" x0.1
31021   Current Phase 3   "A" x0.1

31022   Power Phase 1     "W" x1
31023   Power Phase 2     "W" x1
31024   Power Phase 3     "W" x1

31025   Frequency            "Hz" x0.01

31029   Power Phase 1     ? Same as 31022
31030   Power Phase 2     ? Same as 31023
31031   Power Phase 3     ? Same as 31024

Temperatures inverter
31032   Temperature Inv  "°C"  x0.1
31033   Temp. Inner         "°C"  x0.1

Battery:
31034   Voltage Battery   "V"  x0.1
31035   Current Battery   "A"  x0.1
31036   Power Battery     "W"  x1
31037   Temp. Battery     "°C"  x0.1
31038   SoC Battery        "%"  x1

Name:
30004-300011 "H3-8.0-E for example

Cumulative:
32001 PV Energy total "kWh" x0.1
32002 PV Energy daily "kWh" x0.1

32003 Charge Energy total "kWh" x0.1
32004 Charge Energy daily "kWh" x0.1

32007 Discharge Energy total "kWh" x0.1
32008 Discharge Energy daily "kWh" x0.1
 ** No direct grid connection yet, therefore these vlues are missing here"
32022 EPS Energy total "kWh" x0.1
32023 EPS Energy daily "kWh" x0.1

adamelo1, 2022-10-29:
Hello, 
I assume this wont work for the ETH connection as well, right?

rsaemann, 2022-10-30:
> Hello, I assume this wont work for the ETH connection as well, right? As I have H3-10-E connected via eth port to my HA but I struggle with pairing it up

I first tried the same: Bought the LAN adapter and tried to reach via Modbus-LAN, but no success (port 502 not visible). In a facebook post they said that a new firmware would enable this feature but I did not get it from the support. The support said that Modbus over LAN is not intended and that they do not want to give information about the registers to end-users.
So I had to go via RS485-Modbus using a 3€ adapter device to my Raspberry Pi.

As for the H1 inverters the register adress are different between LAN and 485 Modbus transaction. If you manage to get the newest firmware for your H3 then it would be great to see, if the register adresses are the same. I used QModbus to read everything and then compared with the inverter display.

wranglatang, 2022-10-31:
> Cumulative: 32001 PV Energy total "kWh" x0.1 32002 PV Energy daily "kWh" x0.1
> 
> 32003 Charge Energy total "kWh" x0.1 32004 Charge Energy daily "kWh" x0.1
> 
> 32007 Discharge Energy total "kWh" x0.1 32008 Discharge Energy daily "kWh" x0.1 ** No direct grid connection yet, therefore these vlues 32009-32021 are missing here because they are zero. Seems to handle the Grid in and out connection" 32022 EPS Energy total "kWh" x0.1 32023 EPS Energy daily "kWh" x0.1

I am reading data from my hybrid inverter via the lan port and would really like to get the Cumulative data! Happy to do some tests if any has any suggestions as to what the setting for the Slave and Address would be! :)

adamelo1, 2022-10-31:
I have updated FW on 1.33 but I am kinda noob to RPI and modbus for now, @wranglatang would you mind sharing your yaml files for the reading the data via LAN port? It would be really appreciated 🥇

wranglatang, 2022-10-31:
Im not using an esp / serial converter. My Inverter has a lan port that im able to plug straight into. So i just used the configuration that @StealthChesnut has provided in his code.

adamelo1, 2022-10-31:
I also have inverter with LAN port, but I guess you have H1 type? As the configuration from @StealthChesnut does not work for H3 inverters, or am I wrong?

rsaemann, 2022-11-10:
I created a [fork to add support of H3 inverters](https://github.com/rsaemann/HA-FoxESS-H3-Modbus)

markopeter1973, 2023-01-08:
H3 12kw , works with RS485 to Eth Bridge,

williamjeccles, 2023-03-02:
@rsaemann if you can create a new config file for h3 and submit it as a pull request, i'll review and merge.

closed 2023-03-02 by williamjeccles