StealthChesnut/HA-FoxESS-Modbus #62: [Input/Output] Modbus Error: [Invalid Message] No response received, expected at least 2 bytes (0 received) / No Response received from the remote unit/Unable to decode response

Vato9394, 2022-11-23:
I get 2 errors in the home assistant log:
- [Input/Output] Modbus Error: [Invalid Message] No response received, expected at least 2 bytes (0 received)
- [Input/Output] No Response received from the remote unit/Unable to decode response

But the integration is working fine and all my data is filled in.
is this a bug in the integration or do I do something wrong?

StealthChesnut, 2022-12-01:
Hi @Vato9394 
How are you connecting to your inverter?
Are you using Lan or Modbus?

Vato9394, 2022-12-01:
I am using Modbus (specifically a serial to usb)

StealthChesnut, 2022-12-01:
Ok, thanks.

Can you post the version numbers for your inverter firmware please?

Vato9394, 2022-12-01:
1.36

Vato9394, 2022-12-01:
I am using the fork for the h3 tho.
I have an AIO H3 8.0 E

StealthChesnut, 2022-12-01:
@rsaemann - are you able to help?

rsaemann, 2022-12-02:
I am getting the same error. I will check if it is a bad register. Maybe it is just too many sensors read out in the update loop. I used 30 seconds for every register. Maybe that is not enough. 
First guess for troubleshooting: Disable some sensors that you do not need by commenting the lines. Or increase the update interval.

rsaemann, 2022-12-02:
I found for my installation (island mode), that the sensors for the inverter current, battery current and battery voltage are responsible for the error. If I comment them out, no modbus errors are thrown.

If you have only the main grid or both (grid+eps) connected: please report if disabling the sensors work for you.

Vato9394, 2022-12-02:
I have disabled (commented) following sensors:
- Battery Voltage
- Battery Current

Now, I don't get an error thrown.

Thanks for the help!

StealthChesnut, 2022-12-03:
Ok - closing this issue as it seems to be resolve. Thanks for helping out @rsaemann!

closed 2022-12-03 by StealthChesnut

kevinverelst, 2022-12-05:
@Vato9394 Could you help me setup my Integration? I'm seeing the same errors except my integration isn't working. The values stay "unavailable". I'm doing exactly the same except I have a H3-5.0-E inverter. How did you connect your cable to the RS485 port? If I understand correctly, I need pin 1&2?
[image]

Vato9394, 2022-12-06:
Yes, pin 1 and 2 (I got the RS485 to usb, I first tried with the USR W610, but didn't get it to work, I don't know why, maybe my incompetence?) Connect 1 to A and 2 to B, don't switch this.
I also used the fork for the H3 specifically. https://github.com/rsaemann/HA-FoxESS-H3-Modbus
I copied it manually to home assistant, don't use hacs for this (the locations in the yaml code are different if you use hacs, do yourself a favor and just do it manually)
And my firmware is the latest 1.36 (just send foxess support an email, they can update it remotely)

rsaemann, 2022-12-06:
Did you check the Inverter's RS485 Slave ID? The sensors use standard address 247. However due to installation this might have changed. I encountered an accidential reset once that changed the address to 000.

Check the dongle and connection with a tool like [GModbus](https://www.gineers.com/en/products_soft_other_gmodbus.php) for Windows.

Vato9394, 2022-12-06:
You can check this in the menu on the inverter itself.

tagware, 2022-12-06:
Guys, Girls & Others,

Is there a reason why you guys are using the RS485 port? Rather than plugin
a CAT5(RJ45) cable into the inverter Ethernet port (F) and running it back
to the router?

As a CAT5 shield cable run can be up to 100 meters and still be in spec.

*Save Electrons. :o)*

Regards
*David Vincent*

Vato9394, 2022-12-06:
> Guys, Girls & Others, Is there a reason why you guys are using the RS485 port? Rather than plugin a CAT5(RJ45) cable into the inverter Ethernet port (F) and running it back to the router? As a CAT5 [...]

on the H3 inverters, modbus over ethernet is not (yet) enabled. thats the reason we use the RS485 port.

kevinverelst, 2022-12-06:
@Vato9394 I'm an idiot. It was working from the start but I was looking at the foxess entities (I used to have the foxess cloud integration installed) instead of the modbus entities. I also was under the impression that it was not working due to the error in the logs and just looking at the wrong entities. I cleaned everything up and started from scratch. This is when I noticed that it was working. 

@rsaemann the default address was working fine and I can confirm your fork is working on a H3-5.0-E inverter.

@tagware My inverter has a USB wifi dongle connected to port F. I would rather just plug in an ethernet cable but since the [fork](https://github.com/rsaemann/HA-FoxESS-H3-Modbus) I'm using tells me that port 502 is closed. It seems I cannot connect over network. I'm also confused it says I can connect LAN on port F while that has a USB interface.
[image]

tagware, 2022-12-06:
Okay,

You have a different model to my AC1-5.0-E which does. Cost saving again I
suppose. :o(

But you do have 3 x more PV inputs that I do. As I have none. :o(

Oversight on my part, ended up buying a Solax 2K inverter to make up on
that mistake.

Regards
*David Vincent*

tagware, 2022-12-06:
Hi Kevin,

You in theory can swap over the WiFi with a LAN port for £67 ish. The
trouble is finding one.

Regards
*David Vincent*

kevinverelst, 2022-12-06:
Why bother? The USB RS485 stick I bought was € 3...

tagware, 2022-12-06:
Yep. Not saying you should.

 It was for others to know that you can. As an alternative.

Cheers.

tagware, 2022-12-06:
> Do you mean the FoxESS Smart LAN module? I think this just replaces your
> WiFi Cloud connection with hardwired, rather than allowing you local access.

That's a Ho Hum point.

Yep. I concede.

Stick to you RS485

Regards
*David Vincent*

StealthChesnut, 2022-12-06:
> @Vato9394 I'm an idiot. It was working from the start but I was looking at the foxess entities (I used to have the foxess cloud integration installed) instead of the modbus entities. I also was [...]

@kevinverelst, @Vato9394 - are your problems solved? If so I will close this discussion. :)

Vato9394, 2022-12-06:
yes, mine was resolved some time ago, you can close this discussion ;)