StealthChesnut/HA-FoxESS-Modbus #64: Tried to set this up but keep getting errors

Thebig24fan, 2022-12-01:
Tried setting this up and put ip that’s on inverter but keep’s  failing to connect, how do I check its real ip  ? I’ve tried connecting laptop to same lan lead and getting 50 meg  bb so not as connecting problem  even followed the YouTube video done today no luck spent countless hours trying to get this to work please help thanks 
Error message I get is 
Logger: pymodbus.client.sync
Source: components/modbus/modbus.py:386 
First occurred: 19:51:35 (251 occurrences) 
Last logged: 19:57:59

Connection to (192.168.0.043, 502) failed: timed out
Connection to (192.168.0.043, 502) failed: [Errno 113] Host is unreachable

And  at startup 

Logger: homeassistant.components.integration.sensor
Source: components/integration/sensor.py:184 
Integration: integration (documentation, issues) 
First occurred: 19:51:27 (8 occurrences) 
Last logged: 19:51:27

sensor.load_sum could not restore last state unknown: [<class 'decimal.ConversionSyntax'>]
sensor.bat_charge_sum could not restore last state unknown: [<class 'decimal.ConversionSyntax'>]
sensor.bat_discharge_sum could not restore last state unknown: [<class 'decimal.ConversionSyntax'>]
sensor.feedin_sum could not restore last state unknown: [<class 'decimal.ConversionSyntax'>]
sensor.grid_consumption_sum could not restore last state unknown: [<class 'decimal.ConversionSyntax'>]

Running Home Assistant 2022.11.5
Supervisor 2022.11.2
Operating System 9.3
Frontend 20221108.0 - latest

but had issues been trying since 2022 11 ish  tried to restore back to 10.5 but won’t restore properly so can’t go back before 11.2 but even had problems on 10.5 on both modbus and fox cloud hacs add on

tagware, 2022-12-02:
Morning,

You might want to check the IP number. As the format normally doesn't have
the 0 in front of the 43 and might upset IP stack.

192.168.0.043 should be 192.168.0.43

In fact if you use PING command at the MSDOS CMD prompt. It will not see an
active device when you add 192.168.0.043 instead of 192.168.0.43 :o)

You may have been mislead as the actual Fox ESS invertor control panel
displays 192.168.000.043.

If you use 192.168.000.26 that works. Computers !

Regards
*David Vincent*
*Mobile:* *07785 114965*

*Website: www.tagware.co.uk <http://www.tagware.co.uk/> *
*Blog:     **blog.tagware.co.uk* <http://tagware.info/>

<http://www.tagware.co.uk/>   *13 Sandersfield Road, **Banstead, **Surrey,**SM7
2DH. * *Registered No: 03006925 England. ** - *   *wacky.battle.blaze
<https://map.what3words.com/wacky.battle.blaze> *

This e-mail contains confidential information and is for the exclusive use
of the addressee's. If you are not the addressee, then any distribution,
copying or use of this e-mail is prohibited. If received in error, please
advise the sender and delete it immediately. We accept no liability for any
loss or damage suffered by any person arising from the use of this e-mail.
How we use your data can be viewed here
<https://tagware.co.uk/privacy-policy/>

Thebig24fan, 2022-12-02:
Ok thanks I’ll try that . Tried 192.168.000.043 and still got same response so thought not seen ip 3 000 before I after getting no response I’ll try 0

<https://map.what3words.com/wacky.battle.blaze> *

<https://tagware.co.uk/privacy-policy/>

Thebig24fan, 2022-12-02:
Wow got it working thanks a lot

From the desk of Nickie

<https://map.what3words.com/wacky.battle.blaze> *

<https://tagware.co.uk/privacy-policy/>

tagware, 2022-12-02:
Great to hear.  :o)

Regards
*David Vincent*
*Mobile:* *07785 114965*

*Website: www.tagware.co.uk <http://www.tagware.co.uk/> *
*Blog:     **blog.tagware.co.uk* <http://tagware.info/>

<http://www.tagware.co.uk/>   *13 Sandersfield Road, **Banstead, **Surrey,**SM7
2DH. * *Registered No: 03006925 England. ** - *   *wacky.battle.blaze
<https://map.what3words.com/wacky.battle.blaze> *

This e-mail contains confidential information and is for the exclusive use
of the addressee's. If you are not the addressee, then any distribution,
copying or use of this e-mail is prohibited. If received in error, please
advise the sender and delete it immediately. We accept no liability for any
loss or damage suffered by any person arising from the use of this e-mail.
How we use your data can be viewed here
<https://tagware.co.uk/privacy-policy/>

Thebig24fan, 2022-12-02:
thanks for your swift and great help tried for about 2 weeks and about 12 + hours to get this to work with no joy untill you found the error that i couldn't find the solution too .  all i need now is to get ha onto my laptop hard drive instead of on usb another thing spent hours with no joy even after following lots of how to guides and viedos all get odd issues.

tagware, 2022-12-02:
Is that a request for me to provide links to how to do that?

Basically you run an Virtual environment on the laptop.

https://www.virtualbox.org/

Oracle VBox. You can find YouTube on installing HA on VirtualBox. Then,
once you have the base HA installed and working on the laptop.

Crank up the original version on USB and do a full Backup via HA. Then,
move that backup file on to the Laptop Hard disk. You can normally download
the file using HA back utility.

Then launch the HA vanilla version on the laptop. i.e the VBox version.
Then using the file you copied to the laptop hard drive, restore the backup
you made of the USB version back to the HA on the virtual HA running on the
Laptop.

Then restart the HA system on the Virtual Box system and it should be where
you were on the USB version.

Simple's as a furry friend says. 🤔

Thebig24fan, 2022-12-03:
Final got it work thanks

From the desk of Nickie

https://www.virtualbox.org/

Thebig24fan, 2022-12-03:
Used virtual box for my first taster of home assistant but had lots of issues with it like the 24 hours sleep bug and alot of issues then tried to get it on hard drive of unused laptop and had lots of inconsistent issues and would crash alot and kept having to keep restore backups etc then was shown how to run from usb worked for last 2 weeks but was getting slower and watched video yesterday and finally got it to install all the way on hard disk and onboard ok after 15-20 hours of hassle , now it finally work as expected with no speed  issues or hopefully no stability issues!  Got dashboards setup how I want and next job is installing long ethnic cable and set it up properly as it's on an WiFi extender at moment just to test see if fox modbus working ok and at the moment thanks to you and others it working great !
 Thanks to the community for all the help !
Next job somebody smart is create a bootable code to install home assistant on formatted hard drive so you don't need to do live Linux step that i had lot of problems with and has anyone automated the switching of the work modes of the fox invter at sun set on to feed in and at sun rise change to self use ? Would be big help if anybody know how ?  Thanks

From the desk of Nickie

https://www.virtualbox.org/

tagware, 2022-12-03:
Obviously a cat/rj45 cable is the most stable.

However, if you can't make this work. Then using one of these Power
adapters gives both a very near/strong WiFi point or CAT/RJ45 cable
connection. This assumes that you have a near power point to the inverter.

Here is one I've used. Not promoting it. Just used it.

Link AV600 Powerline Adapter Wi-Fi Kit, Wi-Fi Booster/Hotspot/ Extender,
Wi-Fi Speed up to 300Mbps, 2+1 Ethernet Ports, No Configuration Required,
Wi-Fi Auto-Sync, UK Plug (TL-WPA4220 KIT) £33.

tagware, 2022-12-03:
> Tried setting this up and put ip that’s on inverter but keep’s failing to connect, how do I check its real ip ? I’ve tried connecting laptop to same lan lead and getting 50 meg bb so not as [...]

For those that make it here. It's because he was using invalid IP address. 192.168.0.048 should be 192.168.0.48 :o)

Thebig24fan, 2022-12-03:
> 

Yer tried good quality power line plugs not reliable enough connection  for us as not got many sockets about we’re tried them  got big Ethernet cable delivered today spent most of the afternoon fitting it and works a treat thanks all  just doing tweaks to ha now ie dashboards etc  and would love work mode changes in ha  so would like to automate then on sun rise sun set and interested what everyone else done with  ha and  or with fox modbus ?   Thanks

Thebig24fan, 2022-12-04:
Yer sometimes just a little detail can mess up all your hard work and until you find out what is you may as well be 1000 mile away therefore help from other people makes a great community and really can help people’s experience ! Great  help thanks all and I hope I can help others how others have helped me !

From the desk of Nickie

closed 2022-12-06 by Thebig24fan