StealthChesnut/HA-FoxESS-Modbus #71: Missing system losses sum

FozzieUK, 2022-12-24:
I was asked to have a look at an issue where the daily system losses was zero, but the system losses was correctly showing the current system losses. 

The sensor logic appears to be sensor.system_losses is real time loss, this gets integrated by sensor.losses_sum and then the sensor.losses_daily utility meter (which is zero) is the daily summation.

Having looked at the sensor code added to the configuration.yaml, it seems the sensor.losses_sum definition is missing, if you add it in the daily losses sensor works ok - not sure whether this is the intended method but it does work.

The sensor I added is -

`  - method: left
    name: losses_sum
    platform: integration
    round: 2
    source: sensor.system_losses
    unit_time: h
`

Daibutt, 2022-12-24:
Yep, that's how it was originally intended.
https://github.com/StealthChesnut/HA-FoxESS-Modbus/discussions/9 Calculating system losses

Not sure why it's not in the config yaml code though.....

Ndrinta, 2023-02-18:
Up for you, today I was updating my config file to catch up with the new version here and it's the first thing I noticed the integration sensor for the system losses is missing. I'll request a push for that.

```
  - platform: integration
    name: losses_sum
    source: sensor.system_losses
    unit_time: h
    method: left
    round: 2
```

williamjeccles, 2023-03-02:
I've merged the change from @Ndrinta, all sorted now

closed 2023-03-02 by williamjeccles