StealthChesnut/HA-FoxESS-Modbus #74: Integration causes inverter to crash with "BMS Ext Fault"

IB-UK, 2023-01-09:
I have an occurring fault with my inverter that has only started since my HA modbus connection using ethernet.  I disabled the integration and no inverter issues or reboots in a month.  Re-added the integration on a clean HA install and within a few days I get the same error 

I'm using the standard HA settings but I have removed any sensors I do not use.  I noticed in the Fox app that the battery goes offline and then a couple of minutes later it seems the inverter resets itself.  This is noticeable for me as I have EPS setup on critical circuits and so the inverter resetting switches out the internal relay meaning these circuits lose power for 5-10 seconds.

The firmware versions are
Master : 1.48
Slave : 1.02
Manager : 1.45

tagware, 2023-01-09:
Yep. Sure going to be a READ only software.

100's are using it and don't have a problem?

I would start with the following;

Did you make up the CAT5 cable? Or was it purchased?

Is it running a long way next to other power cables? Is the CAT5 shielded?

Are the EPS circuit ground by what means?

As you could be creating a ground fault via the network switch when you are
connected to the inverter. If the EPS is not correctly installed.

If the EPS circuits are isolated does it still reboot? With the network
cable connected?

The list goes on and on.....

It very unlikely to be software reading the register's.

You could use RS485 option instead? Or stick with the APP provided.

If you upgrade to 1.56 manager which will stop the Ethernet cable working
and still reboots then it's not the software. It's a grounding issue.

Plenty to get on with....

IB-UK, 2023-01-09:
Thanks for the quick response - I have tried to eliminate the obvious - whilst it is READ only software it does still however poll the inverter - I'm not for one minute suggesting it might be a bug in the integration, more likely it is something the inverter doesn't like about the polling.  The fact that the error is a comms fault to the BMS, it suggests that the Fox software maybe can't keep up and the polling of the registers 'could' be causing this.  This might be why they disabled ethernet...  it could be the added load of supporting ethernet.

100s might be using without a problem, likewise if they are not using the internal EPS relay then it is highly likely they will not notice a 5 - 10 second reset - the error is only logged on the inverter itself, not on the cloud or app.

It is a commercial patch cable to a managed hub, which does not run close to other sources of interference.

EPS circuits are grounded by earth rod separate to the grid earth. 

When I tried troubleshooting I simply removed the HA integration, the ethernet remained connected which eliminates the chance of any ground faults as you suggest.

> It very unlikely to be software reading the register's. **I agree it seems implausible, however it is the ONLY variable that triggers these occasional reboots**

> You could use RS485 option instead? Or stick with the APP provided.  **This may be an option I go with, or at least trial to see if it resolves the problem**

**Can I just reiterate that your work on the integration is superb and I am not being critical of it at all - simply attempting to find the solution to a bug.  Please don't view this as an attack on your great work!**

IB-UK, 2023-01-09:
> Don't panic I'm not the developer. I just another bod using the same
software.  - **Oh yeah, I just immediately thought I'd upset the dev, which was not my intention**

I've got spare Cat5 cabling between the inverter location and HA location so I'll just run RS485 and see what happens if it does another reboot.

djdarlo, 2023-01-19:
Any update on this one as I'm seeing the same issues.  I have...

Master 1.46
Slave 1.02
Manager 1.41

IB-UK, 2023-01-20:
> Any update on this one as I'm seeing the same issues. I have...
> 
> Master 1.46 Slave 1.02 Manager 1.41

I'm so glad someone else is experiencing this (not that I'm glad you're having issues).  I was beginning to think it was only me.

Anyway I've moved over to directly using the RS485 to a USB RS485 adaptor on my RaspberryPi and it's been 5 days so far without issue....

I'm wondering if that is why they disabled the LAN port in later firmware versions.  The manual does mention RS485 to allow monitoring etc but doesn't say the same about LAN.

djdarlo, 2023-01-20:
Yep - definitely not just you.  I managed to replicate the crash a few times yesterday.  I've got a RS485 converter on order so will let you know how I get on.

IB-UK, 2023-01-20:
That's interesting - did you manage to make it crash almost on demand

djdarlo, 2023-01-20:
15-20 minutes of connection made it crash.

Jooeee, 2023-01-20:
I've also got a similar issue. My inverter freezes for a couple of minutes and then resets itself but doesn't log any errors. 

After unplugging the ethernet it runs fine.

Inverter Model : H1-5.0-E
Master : 1.44
Slave : 1.02
Manager : 1.39

IB-UK, 2023-01-20:
Last time mine crashed it logged a 'blank' error.  I just upgraded to a V2 Master for the Cube battery so that might be why the error is blank maybe?

So there is definitely something wrong with the inverter using ethernet

spmaid, 2023-01-20:
Just to add to this, I have H1-6.0-E Master : 1.48Slave : 1.02Manager : 1.47 and BMS on 1.12 with batteries on 1.10

I am using ethernet for MODBUS with zero issues as reported above.  

Maybe that might help, but thought I'd add just in case.

Rill189, 2023-02-15:
I had simalr issues... Not all the time just seemed to be when I had high PV power and when I switched a high load on such as a kettle it would cause the entire system to crash with a BMS Volt fault.

I have got hold of Fox ESS. They upgraded the Invertor while they new I used HA. Stating they don't support HA.

They upgraded the Firmware and now my ethernet port is off. I cannot downgrade as I have no access to the firmware packages.

Please help.

williamjeccles, 2023-03-06:
Closing until someone can confirm if this issue was firmware related or integration related.

closed 2023-03-06 by williamjeccles