StealthChesnut/HA-FoxESS-Modbus #78: WiFi / LAN Feed sending no data

sofapi, 2023-01-13:
Had a brand new system installed today and was quite looking forward to just cracking on with the HA build and better integration direct without relying on FoxESS Cloud and their slow polling. 

Anyone had any joy extracting sensor data from the network port on the most recent firmware?  I can't understand why a vendor would disable or limit this interaction?!

tagware, 2023-01-13:
Welcome to the club.

I had to put the manager back to version 1.50 to get the LAN port back.

Report it as a Bug. But, received no explanation of what or why.

😭

sofapi, 2023-01-13:
> Welcome to the club. I had to put the manager back to version 1.50 to get the LAN port back. Report it as a Bug. But, received no explanation of what or why. 😭
> […](#)

Weirdly enough, mine is showing:
Manager version 1.17.

where do you download the manager versions, is it a usb firmware flash job? 

Cheers for the advice.

tagware, 2023-01-13:
Hi

My system is AC1-5.0-E system

So, milage may vary. 🤔

You can normally apply the firmware from a Agent account on the Fox Portal.

Or ask Fox support or your installer to do it for you? As getting it wrong
can stop the inverter from working.😭

I happened to have worked in the IT industry for 40 odd years.🥳 Hence are
quite use to doing this stuff. The portal has lots of choices on firmware.
🥸

Alternative, a number of HA options do interaction with the portal. Or,
failing that you can use RS485 to WiFi option.

Which seems to be the route that is unlikely to be removed. But, that is
what I thought about the LAN option. 🤔

tagware, 2023-01-13:
The LAN only had a short time period of working correctly. 1.41 to 1.50 on
my case.

So, upgrade might do the trick.

vladimirus, 2023-01-30:
@tagware 

> The portal has lots of choices on firmware.

Would you be able to point me where in the portal you can update? I am registered as "installed" and I can't find anything of sorts. Thanks in advance

williamjeccles, 2023-02-02:
> @tagware
> 
> > The portal has lots of choices on firmware.
> 
> Would you be able to point me where in the portal you can update? I am registered as "installer" and I can't find anything of sorts. Thanks in advance

You need to be an agent profile type for firmware management.

closed 2023-02-02 by williamjeccles