StealthChesnut/HA-FoxESS-Modbus #79: Question about sensor update frequency

benbeton, 2023-01-17:
I have set my sensors with a scan_interval of 5 - 30 sec. However, the values update very differently, often there are some minutes no update, sometimes even up to 25 min. although the values change with the cloud integration very fast. 

But since I prefer the solution via Modbus, I would like to find out if this is a normal behavior or if it can be due to a misconfiguration. Here are a few examples. In this case, there was bright sunshine and the battery was charging at about 3 - 4 KW, so the SoC sensor should have changed promptly, but instead there are jumps of ~20%. 

**My setup**
- FoxESS Inverter H3 6.0E 6.5KW Battery
- USR-610 Modbus LAN converter

**SoC charge state screenshots:**
[image]
[image]
[image]

**SoC sensor config:**
```
- name: "fox_bat_soc"
    scan_interval: 30
    slave: 247
    address: 31038
    state_class: measurement
    unit_of_measurement: "%"
    data_type: uint16
    input_type: holding
    device_class: battery
```

**Modbus config**
```
- name: FoxESS_Inverter
  type: tcp
  host: 192.168.XX.XXX
  port: 502
  retry_on_empty: true
  retries: 3
  timeout: 5
  close_comm_on_error: false
```

in the same time template sensor (pv1_power & pv2_power) has delivered timely data:

[image]
[image]

so it looks like the inverter is providing data but very different in time, is there any explanation or way to change this?
I would like to have as accurate data as possible

regards

williamjeccles, 2023-02-02:
What poll rate do you have set on the USR converter? Likely HA is asking the USR for updates every 30 seconds but the modbus settings on the USR are set with a slower polling rate.

benbeton, 2023-02-03:
this is my current polling config on the USR:
[image]

closed 2023-02-13 by benbeton