StealthChesnut/HA-FoxESS-Modbus #93: Time sensors appear to be missing ?

SteveBrailsord, 2023-03-03:
Hi
These sensors appear to be missing  are missing - where should they be obtained from please ?
sensor.time_period_1_start
sensor.time_period_1_end
sensor.time_period_2_start
sensor.time_period_2_end

They are used below :-
- name: "Time Period 1 - Start Time"
        icon: "mdi:calendar-clock"
        state: >
          {% set raw_time = states('sensor.time_period_1_start') | int %}
          {% set hours = raw_time // 256 %}
          {% set minutes = raw_time - (hours * 256) %}
          {{ strptime(hours | string + ":" + minutes | string, '%H:%M').time() }}
      - name: "Time Period 1 - End Time"
        icon: "mdi:calendar-clock"
        state: >
          {% set raw_time = states('sensor.time_period_1_end') | int %}
          {% set hours = raw_time // 256 %}
          {% set minutes = raw_time - (hours * 256) %}
          {{ strptime(hours | string + ":" + minutes | string, '%H:%M').time() }}
      - name: "Time Period 2 - Start Time"
        icon: "mdi:calendar-clock"
        state: >
          {% set raw_time = states('sensor.time_period_2_start') | int %}
          {% set hours = raw_time // 256 %}
          {% set minutes = raw_time - (hours * 256) %}
          {{ strptime(hours | string + ":" + minutes | string, '%H:%M').time() }}
      - name: "Time Period 2 - End Time"
        icon: "mdi:calendar-clock"
        state: >
          {% set raw_time = states('sensor.time_period_2_end') | int %}
          {% set hours = raw_time // 256 %}
          {% set minutes = raw_time - (hours * 256) %}
          {{ strptime(hours | string + ":" + minutes | string, '%H:%M').time() }}

rand12345, 2023-03-05:
This was bugging me for a while

Edit your Lovelace dashboard, select the Time Period 1 Start and type "time". A second sensor will appear with the decoded time. Rinse and repeat for the other sensors.

If they don't show up, ensure you are including the custom_components/HA-FoxESS-Modbus/configuration.yaml file and not modbusLAN.yaml or modbusUSB.yaml in your main configuration.yaml file.

SteveBrailsord, 2023-03-05:
Thanks
That sorts it
First time ever - no watchman errors and no log errors 😊
I’m a happy man !

closed 2023-03-06 by williamjeccles