StealthChesnut/HA-FoxESS-Modbus #94: address: 11038 dupliate

iancg, 2023-03-05:
I noticed the following in the HA log:

2023-03-05 13:12:48.334 WARNING (MainThread) [homeassistant.components.modbus.validators] Modbus sensor/Outer-BMS Cell Temp Reported address 11038_input_247 is duplicate, second entry not loaded!

And sure enough it is in there twice

    - name: "Inner-Battery-Temp"
      scan_interval: 60
      slave: 246
      address: 11038
      state_class: measurement
      unit_of_measurement: "..C"
      scale: 0.1
      precision: 1
      data_type: int16
      input_type: input
      device_class: temperature

and

    - name: "Outer-BMS Cell Temp Reported"
      scan_interval: 30
      slave: 247
      address: 11038
      state_class: measurement
      unit_of_measurement: "..C"
      data_type: int16
      scale: 0.1
      precision: 1
      device_class: temperature
      input_type: input

I checked the wiki, but couldn't see what the error is, but I wonder if (based on the pattern) if the second one (Outer-BMS Cell Temp Reported) should be 11049?

rand12345, 2023-03-06:
@williamjeccles 

There's overlap with mature sensors and the newer ones. Do you want to make a decision about how these are labelled before dropping a duplicate?

williamjeccles, 2023-03-06:
I'll modify it now, thanks for the report. https://github.com/StealthChesnut/HA-FoxESS-Modbus/pull/97

closed 2023-03-06 by williamjeccles