StealthChesnut/HA-FoxESS-Modbus #102: Prototype layout for repo

rand12345, 2023-03-06:
I noticed that you are integrating H3 inverters into the repo. Having seen this form of integration before with other platforms I'm offering a suggestion for a new layout which should help the maintainer's sanity when propagating changes. This isn't a PR although you are welcome to take it.

Pros: 

- User IP configuration is outside the repo, all held in /secrets.yaml
- Selecting a communications method is set in /configuration.yaml and users to not edit repo files
- Better maintainability, common modbus entries are shared between communication methods. One change propagates through inheritance
- Feature updates require no selective copy and pasting which nips yaml issues in the bud
- One copy and paste entry is needed to add this config to /configuration.yaml (see edited readme.md)
- Can be symlinked to /config/custom_componants/HA-FoxESS-Modbus permitting git pulls within the home directory

Cons:

- Breaking change occurs through instantiation, needs configuration.yaml updating and two entries in /secrets.yaml

https://github.com/rand12345/HA-FoxESS-Modbus/tree/new-layout-prototype

Ndrinta, 2023-03-08:
Up for you.
I remember finding extremely confusing to have to use the "USB" version to then edit it to change it use he USR even thought that's on LAN so why wouldn't I use the "LAN" one? After reading the documentation then started to make sense but I never accepted the naming convention... I thought, wouldn't be better to have the "USB" version renamed to "RS485" and then you choose between USB or LAN?
May be better to have a pure LAN and 2 flavours of RS485 (RS485 -USB and RS485-LAN). Agree, still a little confusing but at least I can just copy and paste the code from git without making changes at every update.
But I believe your solution is far better.

williamjeccles, 2023-03-08:
I'm discussing with Chris about adding this as a user setting so you can simply install this integration via Hacs, during the install we ask you to select your inverter model and connection method, provide an IP address and bingo it's now working and future updates can be auto-applied.