StealthChesnut/HA-FoxESS-Modbus #119: KH series inverters

TonyM1958, 2023-04-04:
Based on recent experience, I think you could add KH single phase inverters to the list as an enhancement with a wiki page on setting this up?

Heres the thing: there is no ethernet port so you connect to RS485 via the CT connector. You hook up W610 to this and scan for modbus ports and you find 31000!

So, surprisingly, KH then just works with the modbusLAN sensor configuration and the IP address of W610.

You would need a USB serial port config in modbusLAN for it to work with a USB to RS485 adapter.

TonyM1958, 2023-04-05:
One other note on this. If the LAN profile / 310xx is used for RS485 over USB on KH, does it need all the sensors to include slave_address 247?

dirkesquire, 2023-10-14:
Hello, I have a K Series - KH10. Is this supported? I came to the Issues to work out why it is not included on the supported hardware list. Thank you!

https://www.fox-ess.com/1-phase/
https://www.fox-ess.com/download/upfiles/EN-K-Series-Datasheet-V1.7-7.7.pdf

TonyM1958, 2023-10-14:
This integration is no longer being developed. We have working integrations for KH being used by others in SA.

Try this: https://github.com/nathanmarlor/foxess_modbus

Or this: https://github.com/TonyM1958/HA-FoxESS-Modbus

closed 2023-10-14 by TonyM1958