StealthChesnut/HA-FoxESS-Modbus #120: Scan interval

TonyM1958, 2023-04-05:
Just put a vote in for a review of scan interval settings?

I use 30 seconds for most sensors, with 5 seconds for power sensors.

For example, do we need to scan BMS cell mV high / low  every 5 seconds, would 30 seconds suffice?

Should there be !secret scan_interval setting so people can make a global change up or down to the default 30 seconds if they wish?

TonyM1958, 2023-04-05:
For info, here's an example of what I use to allow the scan_interval to be customised more easily:

In secrets.yaml:

```
# sensor scans - high = 30 seconds, medium = 1 minute, low = 5 minutes
scan_interval_high: 30
scan_interval_medium: 60
scan_interval_low: 300
```

In modbusUSB.yaml:

```
    - name: "PV1 Voltage"
      scan_interval: !secret scan_interval_high
      slave: 247
      address: 11000
...
```

You can always over-ride an individual sensor by putting a fixed value directly in the file.

canton7, 2023-04-16:
Be very very careful here.

If you're using th W610 (and a lot of people are), that's only capable of reading one register every 700ms or so. However, even the default scan_intervals add up to more than this, which means that read tasks just back up inside home assistant (there's no logic to say "Am I trying to do a read already? OK, I won't try again until this is complete").

The moment you start using scan_intervals of 1 or so, you really run up against the limits of the W610. If you even have two sensors with a scan_interval of 1, and no other sensors, that's beyond what the W610 can handle, and tasks are jut going to be backing up in home assistant. You can have ~6 sensors with a scan_interval of 5, and no other sensors at all. And so on.

Unfortunately there's no logging by default to tell you what's happening.